raphaelthurnherr/algobot_onionOmega #4: Continous FRP

YannCaron, 2018-05-31:
- Create a new topic (Stream) that send periodically all the sensors informations.
- Create a new command that activate it (ActivateStream for example).

raphaelthurnherr, 2018-06-08:
Add a topic "Stream" to send periodically the status message with all data.
Actually the Stream is sended all 100mS.

`{
    "MsgTo": "unknown",
    "MsgFrom": "algo_98f8",
    "MsgID": 0,
    "MsgData": {
        "MsgType": "event",
        "MsgParam": "status",
        "MsgValue": [
            {
                "name": "algo_98f8",
                "upTime": 0,
                "firmwareVersion": "18.06.02",
                "mcuVersion": "10",
                "boardRev": "17",
                "battery_mv": 4181
            },
            {
                "din": 0,
                "state": 0
            },
            {
                "din": 1,
                "state": 0
            },
            {
                "din": 2,
                "state": 0
            },
            {
                "din": 3,
                "state": 0
            },
            {
                "btn": 0,
                "state": 0
            },
            {
                "btn": 1,
                "state": 0
            },
            {
                "motor": 0,
                "cm": 0,
                "speed": 0
            },
            {
                "motor": 0,
                "cm": 0,
                "speed": 0
            },
            {
                "sonar": 0,
                "cm": 97
            },
            {
                "pwm": 0,
                "state": 0,
                "power": 0
            },
            {
                "pwm": 1,
                "state": 0,
                "power": 0
            },
            {
                "pwm": 2,
                "state": 0,
                "power": 0
            },
            {
                "pwm": 3,
                "state": 0,
                "power": 0
            },
            {
                "pwm": 4,
                "state": 0,
                "power": 0
            },
            {
                "pwm": 5,
                "state": 0,
                "power": 0
            },
            {
                "pwm": 6,
                "state": 0,
                "power": 0
            },
            {
                "pwm": 7,
                "state": 0,
                "power": 0
            },
            {
                "pwm": 8,
                "state": 0,
                "power": 0
            },
            {
                "pwm": 9,
                "state": 0,
                "power": 0
            }
        ]
    }
}`

closed 2018-06-08 by raphaelthurnherr

YannCaron, 2018-06-08:
That's perfect dude ! So many thank you ;-)