raphaelthurnherr/algobot_onionOmega #6: Application update

raphaelthurnherr, 2018-06-01:
System command:  firmware update, restart, reboot

{
   "MsgID" : 0,
   "MsgTo" : "algo_2643",
   "MsgFrom" : "MQTTFX",
   "MsgData" : {
      "MsgType" : "command",
      "MsgParam" : "system",
      "MsgValue": [
         {
            "update": "now"     // "auto" (at application startup)
         }, 
         {
            "reboot": "now"     //system reboot
         }, 
         {
            "restart": "now"     // application restart
         }
      ]
   }
}

YannCaron, 2018-06-01:
Génial !

raphaelthurnherr, 2018-08-01:
For update check:
..
  "MsgData": {
    "MsgType": "command",
    "MsgParam": "system",
    "MsgValue": [
      {
        "application": "check"
      }
    ]
  }
..

for application update:
..
  "MsgData": {
    "MsgType": "command",
    "MsgParam": "system",
    "MsgValue": [
      {
        "application": "update"
      }
    ]
  }
..

closed 2018-08-01 by raphaelthurnherr