raphaelthurnherr/algobot_onionOmega #9: Continuous FRP optimization

YannCaron, 2018-06-08:
Prefer use an associative array (key value pair) than a list for better performances.

the actual message is the following :
```json
{
    // ....
        "MsgValue": [
            {
                "name": "algo_98f8",
                "upTime": 0,
                "firmwareVersion": "18.06.02",
                "mcuVersion": "10",
                "boardRev": "17",
                "battery_mv": 4181
            },
            {
                "din": 0,
                "state": 0
            },
            {
                "din": 1,
                "state": 0
            }
    // ....
}
```

It is better to use associative array / map for msg values.
This is the expected message :

```json
{
    // ....
        "MsgValue": { // be careful the previous symbol as changed from [ to {
            "data" : {
                "name": "algo_98f8",
                "upTime": 0,
                "firmwareVersion": "18.06.02",
                "mcuVersion": "10",
                "boardRev": "17",
                "battery_mv": 4181
            },
            "din0" : 0,
            "din1": 1,
            "motor0" : {
                "cm": 0,
                "speed": 0
            },
    // ....
}
```

then I can access to data as the following :
```javascript
var speed = MsgData.MsgValue.motor0.speed;
```

for example

raphaelthurnherr, 2018-06-08:
Aie... ca me fait des changements de ouf...

YannCaron, 2018-06-08:
Oui, mais point de vue algorithme, ça n'a rien à voir.
Actuel : je boucle sur tous les status à chaque fois
Map : javascript se charge pour moi de mettre dans un hash map, donc acces
direct

raphaelthurnherr, 2018-06-13:
This is the result for your request !
Please update your firmware and let me know if it's ok !

```
"MsgTo":"MQTTFX",
"MsgFrom":"algo_2643",
"MsgID":14351139,
"MsgData":{
"MsgType":"response",
"MsgParam":"status",
"MsgMap":{
"name":"algo_2643",
"upTime":0,
"firmwareVersion":"18.06.02",
"mcuVersion":"10",
"boardRev":"17",
"battery_mv":0,
"din":[
{
"state":0
},
{
"state":0
},
{
"state":0
},
{
"state":0
}
],
"btn":[
{
"state":0
},
{
"state":0
}
],
"motor":[
{
"cm":0,
"speed":0
},
{
"cm":0,
"speed":0
}
],
"sonar":[
{
"cm":4
}
],
"pwm":[
{
"state":0,
"power":0
},
{
"state":0,
"power":0
},
{
"state":0,
"power":0
},
{
"state":0,
"power":0
},
{
"state":0,
"power":0
},
{
"state":0,
"power":0
},
{
"state":0,
"power":0
},
{
"state":0,
"power":0
},
{
"state":0,
"power":0
},
{
"state":0,
"power":0
}
]
}
}
}
```

closed 2018-06-13 by raphaelthurnherr