raphaelthurnherr/algobot_onionOmega #22: Create an initialize command

YannCaron, 2018-07-04:
A command that reset all states (to be sent at the beginning of program, just after connection).

e.g. Set all leds, motors, pwms to off, reset all captors, distance and so on.

raphaelthurnherr, 2018-08-01:
Use the restart command.

..
   "MsgData":{
      "MsgType":"command",
      "MsgParam":"system",
      "MsgValue":[
         {
            "application":"restart"
         }
      ]
   }
..

raphaelthurnherr, 2018-08-03:
Also possible to use this config message

...
      "MsgValue":[
         {
            **"config":{
               "reset":"true"
            },**
            "stream":{
               "state":"off",
               "time":1000,
               "onEvent":"off"
            }
         }
      ]
...

closed 2018-08-03 by raphaelthurnherr