raphaelthurnherr/algobot_onionOmega #24: Servo message incorrect

YannCaron, 2018-07-04:
When I send :
``` json
{
  "MsgID": 0,
  "MsgTo": "algo_973a",
  "MsgFrom": "algo",
  "MsgData": {
    "MsgType": "command",
    "MsgParam": "servo",
    "MsgValue": [
      {
        "servo": 0,
        "state": "on",
        "angle": 45
      }
    ]
  }
}
```

I receive :
`algo_973a -> Message ID: 0 -> algo_973 -> MESSAGE ALGOID INCORRECT RECU !`

raphaelthurnherr, 2018-07-04:
Need to enable during compilation and to be tested...

raphaelthurnherr, 2018-08-16:
Servomotor are now functionnal on version 1.4.0
Please use this message with param "servo"  and Value with "pwm"

``` json
{
  "MsgID": 0,
  "MsgTo": "algo_973a",
  "MsgFrom": "algo",
  "MsgData": {
    "MsgType": "command",
    "MsgParam": "servo",
    "MsgValue": [
      {
        "pwm": 0,
        "state": "on",
        "position": 100
      }
    ]
  }
}
```

closed 2018-08-16 by raphaelthurnherr